code-iai/ROSIntegration #176: Fork with visualizations, websockets, latest rosbridge_suite, etc.

naxsm, 2022-07-18:
I thought this repo was mostly abandoned so I started working on my own fork to work with the latest rosbridge_suite and add some functionality. I see this repo is now active again and I'd like to contribute my work, but some of it is incompatible with the current state of this repo. Also my fork is still a work in progress and not ready yet.

What's the best way to move forward? I'd like ROSIntegration to be a single package with options to support e.g. tcp or ws, ros1 or ros2, different data formats from rosbridge_suite, etc.

Sanic, 2022-07-20:
Hi @naxsm 
Thanks for considering integrating the functionality together. I think it's also a nice idea for the users to keep everything together.

I gave this a thought and i think there are multiple ways of integrating this. Since the main change seems to be the communication channel, i'd propose to keep the general conversion functionality and everything that directly interacts with Unreal functionality (for example the code for enabling BP usage). So based from your description, i'd assume that the main changes have to be done in the rosbridge2cpp part of this Plugin, which is doing the main communication between UE4 and ROS.

Is using ros2 with rosbridge a major difference in terms of the communication or is it pretty much the same to ros1 and rosbridge? What would needed to be changed there?
For websockets there seems to be some builtin functionality in UE4 (https://docs.unrealengine.com/4.27/en-US/API/Runtime/WebSockets/) that one could use.

tsender, 2022-07-20:
Hi @Sanic and @naxsm ,

I've always wanted a way to work with ROS2 (I currently use ROS2 with this plugin, but I use the ros1_bridge to do the msg/srv conversions, which can be annoying).

There have been some discussion in this issues page: https://github.com/code-iai/ROSIntegration/issues/102

One member @GaoGeolone had modified the rosbridge tcp code to work with this plugin, but RobotWebTools soon thereafter removed all tcp communication from the ros2 branch for rosbridge since ros2 does not utilize tcp. The rosbridge repo only supports websockets now. I did make a fork of this plugin and sort of got a websocket to work (I copied some internal code in UE4's websocket module and modified the rosbridge2cpp portion), but there were still various bugs/issues that needed to be fixed, and I haven't had the time to look into it again.

Some of the issues I ran into with websockets was:
1. The rosbridge node got extremely slow once I added multiple UE4 instances.
2. Within my fork of this plugin, I was unable to add a way to handle the topic queue size, and so my websocket was backed up with trying to publish waaaay too many messages without deleting stale messages. In addition, either the rosbridge websocket or my ROSIntegration websocket was unable to handle subscribing to messages without considerable delay.

naxsm, 2022-08-01:
This is the repo I'm working on right now: https://github.com/naxsm/ROSIntegration. It's not ready for prime time yet, but it works well on ros2 over tcp when using my fork of rosbridge_suite, which adds tcp and fixes some small bugs: https://github.com/naxsm/rosbridge_suite. This one is also a work in progress, and I have yet to synchronize with them, but please feel free to try them and I'll be happy to answer questions and take comments.

tsender, 2022-08-02:
I'm curious as to what visualizations you added to your fork of the plugin. I saw some of code, but no comments, so it was hard to follow.

MCXiaoXu, 2024-12-29:
> This is the repo I'm working on right now: https://github.com/naxsm/ROSIntegration. It's not ready for prime time yet, but it works well on ros2 over tcp when using my fork of rosbridge_suite, which adds tcp and fixes some small bugs: https://github.com/naxsm/rosbridge_suite. This one is also a work in progress, and I have yet to synchronize with them, but please feel free to try them and I'll be happy to answer questions and take comments.
> 
> 
> 这是我现在正在做的repo：[https：//github.com/naxsm/ROSIntegration](https://github.com/naxsm/ROSIntegration)。它还没有准备好，但是当使用我的rosbridge_suite分支时，它在ros2 over tcp上运行得很好，它添加了tcp并修复了一些小错误：[https：//github.com/naxsm/rosbridge_suite](https://github.com/naxsm/rosbridge_suite)。这一个也是一个正在进行的工作，我还没有与他们同步，但请随时尝试他们，我很乐意回答问题，并采取意见。

Hello, I put rosbridge_stuite into the workspace for colcon build and it passed, but I am a newbie and don't know how to run it. Can you help me