code-iai/ROSIntegration #183: PR #180 Gives Compile Error in UE4

tsender, 2022-10-30:
See my comment at the bottom of the PR page: https://github.com/code-iai/ROSIntegration/pull/180

Essentially, the added BP feature uses type 'double', but only UE5 supports type 'double' in blueprints, UE4 does not. Attempting to place the appropriate code in a preprocessor directive to check if the engine version is 4 or 5 does not work, as UE does not support this with UFUNCTIONS (darn).

The only options are to remove the `void OnFloat64Message(const double& Data)` function added in RI/Topic.h and its corresponding cpp code. OR to create a separate branch for UE5.

This is not my repo, so I'm not sure how you want to proceed @Sanic (but my preference is to remove the code, but that's because I much prefer cpp over blueprints).

kabel-knrd, 2022-11-04:
Hi! 
I currently have the same problem with the UE4.27. 
@tsender Have you found a decent workaround that works for you? So have you found a way, where the plugin compiles?

Thanks!

Sanic, 2022-11-04:
Hi @tsender 
Thanks for reporting!
I gave it a thought i think that the main platform of the master branch should be UE4.
So i would be happy to receive a PR for the partial removing of #180 with the related code you've described here.

tsender, 2022-11-08:
Hopefully there aren't too many more issues with differences b/w UE4 and UE5 that can't be easily accounted for, like in preprocessor directives, but you might want to add a note to the README saying the master branch is maintained for UE4, and if the need arises, then perhaps you can make a UE5 branch. Just a thought.

closed 2022-11-08 by tsender