code-iai/ROSIntegration #186: How can I solve thr problem of `sensor_msgs/CameraInfo does not have a field d`

pisces365, 2022-11-16:
I install the plugin of ROSIntegrationVision and I add the VisionActor into the 3D world.
But when I click run,the ros bridge show the error
`[ERROR] [1668582258.304523]: [Client 0] [id: publish:/unreal_ros/camera_info:50] publish: Message type sensor_msgs/CameraInfo does not have a field d`
I do not know where and how can I fix this.
So I wish someone can help me, thanks a lot!

I use Ubuntu 18.04 and UE 4.26.2, here is some image
[image]
[image]

Hemofektik, 2022-11-16:
Looks like the elements are not capital letters in the [serialization code ](https://github.com/code-iai/ROSIntegration/blob/d0018d501203b80fcf4df0b437fa6bc6aed77172/Source/ROSIntegration/Private/Conversion/Messages/sensor_msgs/SensorMsgsCameraInfoConverter.h#L61-L64) as they should be according to the [message definition](https://docs.ros.org/en/lunar/api/sensor_msgs/html/msg/CameraInfo.html).

pisces365, 2022-11-16:
> Looks like the elements are not defined in capital letters in the [serialization code ](https://github.com/code-iai/ROSIntegration/blob/d0018d501203b80fcf4df0b437fa6bc6aed77172/Source/ROSIntegration/Private/Conversion/Messages/sensor_msgs/SensorMsgsCameraInfoConverter.h#L61-L64) as they should be according to the [message definition](https://docs.ros.org/en/lunar/api/sensor_msgs/html/msg/CameraInfo.html).

It seems that the problem has been solved. Thank you very much

closed 2022-11-16 by pisces365