code-iai/ROSIntegration #187: ROS bridge for UE 5

Wangmmstar, 2023-01-03:
Hello, Thank you for your hard work and detailed info. I am currently working on integrating UE5 and ROS. Can this plugin be used for UE5? Or Are there any resources to recommend? Thanks in advance.

Sanic, 2023-01-09:
Hi @Wangmmstar 
Personally, i don't have experience with this plugin and UE5, but it seems that some users here have worked on it:
https://github.com/code-iai/ROSIntegration/issues/143#issuecomment-1278381781

Wangmmstar, 2023-01-10:
Thank you very much@Sanic. I will try the configuration :)

Wangmmstar, 2023-07-10:
> I can also confirm that this plugin is working fine with the Unreal Engine 5.2 binaries. I think this issue can be closed?
Thank you! Will try UE5.2 soon. I close this issue for now.

closed 2023-07-10 by Wangmmstar