code-iai/ROSIntegration #198: /Clock publish rate too low.

rahulranjan29, 2023-08-31:
Hi,

Issue: I'm using the ROSIntegration plugin with 'simulated time = true'. When I check the topic publication rate in ROS via  ```rostopic hz /clock```, I only get 3Hz.  To check if this is caused on the Unreal Side, I calculated the rate at which ```UROSIntegrationGameInstance::OnWorldTickStart(UWorld * World, ELevelTick TickType, float DeltaTime)``` is called and it's ~30Hz.  It looks like messages are being lost over TCP. I'm not sure what is happening.
```
clock: 
  secs: 1693475858
  nsecs: 126977761
---
clock: 
  secs: 1693475858
  nsecs: 460311265
---
clock: 
  secs: 1693475858
  nsecs: 793644673
---
clock: 
  secs: 1693475859
  nsecs: 126978177
---
clock: 
  secs: 1693475859
  nsecs: 460311681
---
clock: 
  secs: 1693475859
  nsecs: 793645089
---
clock: 
  secs: 1693475860
  nsecs: 126978593
---
clock: 
  secs: 1693475860
  nsecs: 460312001
---
clock: 
  secs: 1693475860
  nsecs: 794506977
---
clock: 
  secs: 1693475861
  nsecs: 128380193
```

How can I debug this? Is there a way to increase the clock publication frequency without setting the bfixedUpdateInterval variable?

My Setup:
Vanilla Unreal Engine 5.2 project with ROSIntegration plugin (dev branch)
ROS Noetic on WSL 
 * /rosdistro: noetic
 * /rosversion: 1.16.0

tsender, 2023-09-03:
Hi @rahulranjan29,

You really shouldn't be having such a problem. My advice may be a bit limited as I do not use WSL or UE5. However, based on your post, it seems that you are using the same Windows computer for both UE5 and WSL. One thing you may be experiencing is that by default, UE will run slower if it is not the active window. If this were the case, then you would definitely be seeing a slower game frame rate while you are using WSL to check the /clock topic frequency while UE is running in the background. You can disable this by going to Edit -> Editor Preferences -> General -> Performance, and disable the item pertaining to running in the background (I forget what it's called off the top of my head).

If this does not address the problem, then more information on your setup would be needed.

rahulranjan29, 2023-09-04:
@tsender Unticking "Use less CPU in background" worked wonders!!! Thank you so much! I spent a couple of days trying to solve this and in the end it was this small thing. Ah!

closed 2023-09-04 by rahulranjan29