code-iai/ROSIntegration #200: sensor_msgs/Image Error

jonathanembleyriches, 2023-10-23:
Hi,

I'm currently having an issue with trying to subscribe to a "sensor_msgs/Image" topic. After subscribing I do not receive any callbacks. The only message in the console that could be related is;
`
LogROS: Error: Error on BSON parse - Ignoring message`

This is the following code and I can confirm that the topic is being published and is formatted correctly.
```
SpotFrontLeftCameraTopic= NewObject<UTopic>(UTopic::StaticClass());

SpotFrontLeftCameraTopic->Init(ROSInst->ROSIntegrationCore, TEXT("/spot/camera/frontleft/image"), TEXT("sensor_msgs/Image"));

std::function<void(TSharedPtr<FROSBaseMsg>)> callback = 
        [this](TSharedPtr<FROSBaseMsg> msg) { this->SpotFrontLeftCameraTopicCallback(msg); };
SpotFrontLeftCameraTopic->Subscribe(callback);
```

Do you have any potential solutions or ideas on why this is failing?

Many thanks,
Jon

tsender, 2023-10-27:
Hi @jonathanembleyriches ,

I can tell you this is the line triggering that error message: https://github.com/code-iai/ROSIntegration/blob/46dd9668eab4e5624b88681f22b7303af3d38592/Source/ROSIntegration/Private/rosbridge2cpp/TCPConnection.cpp#L194

I am not quite sure what is causing this to occur on the little info you provided. Can you please provide more details about your  rosbridge setup? Can you subscribe successfully to other types of messages in UE4? Is it just the sensor_msgs/Image message causing a problem? What happens if you lower the camera resolution, will ROSIntegration parse the BSON message then?

Just throwing out some ideas to help narrow down the problem.

jonathanembleyriches, 2023-11-20:
Hi,

I manually changed the ros bridge suite code (not this plugin) which was encoding and sending over TCP. It appears that it was encoding the message twice and therefore creating issues on the receiving end. I'm not entirely sure why it was doing this. Closing this issue as it appears to be due to this. Thanks for your response!

closed 2023-11-20 by jonathanembleyriches