code-iai/ROSIntegration #202: Trigger implementation

brean, 2024-01-14:
I updated to the current ROSIntegration and noticed that this commit introduces an issue that makes it impossible to compile anymore:
https://github.com/code-iai/ROSIntegration/commit/26c6f3fceca013b4204689078d1ed5d02d722585

The issue is that BCON_NEW is not allowed without parameter for the Linux implementation of libbson. However the needed parameter for success and message are added later in [line 41 of StdSrvsTriggerResponseConverter.cpp](https://github.com/code-iai/ROSIntegration/commit/26c6f3fceca013b4204689078d1ed5d02d722585#diff-61239f3b60eb239c53b02d974c2714822d74ca12dc12770b2185fe05cce45767R41)

(see also https://github.com/ros/ros_comm_msgs/blob/kinetic-devel/std_srvs/srv/Trigger.srv )

I think instead of BCON_NEW we need something like BSON_INITIALIZER but I am not familiar enough with libbson to get this working, so maybe someone else can take a look?

tsender, 2024-01-25:
Hello @brean ,

I myself have not yet updated to the latest version, however I can help you fix this problem on your own fork for now. In the one file you mentioned, you can replace the line `*BSONRequest = BCON_NEW();` with `*BSONRequest = bson_new();` I did not write that piece of code, but my suggested fix is how I typically work with the libbson library for this plugin. Let me know if this works.

adarshjamuaar, 2024-01-29:
> Hello @brean ,
>
> I myself have not yet updated to the latest version, however I can help you fix this problem on your own fork for now. In the one file you mentioned, you can replace the line [...]

I was facing the same issue. Replacing it with bson_new() works. Thanks!