code-iai/ROSIntegration #205: ROSBridge Message Error for sensor_msgs/CameraInfo

andreschreiber, 2024-02-03:
I'm running ROS Noetic and Ubuntu 20.04 on a machine running ROS Bridge, and when I try to receive sensor_msgs/CameraInfo data published from UE4, ROSBridge reports an error:
`"publish: Message type sensor_msgs/CameraInfo does not have a field d"`

andreschreiber, 2024-02-03:
Adding this in case anyone else has the issue.
It's due to _bson_append_camera_info having its names for D, K, P, and R being lower-case (they should be uppercase).
Adjusting the code (lines 61-64 on SensorMsgsCameraInfoConverter.h) to be:

```
_bson_append_double_tarray(b, "D", msg->D);
_bson_append_double_tarray(b, "K", msg->K);
_bson_append_double_tarray(b, "R", msg->R);
_bson_append_double_tarray(b, "P", msg->P);	
```
fixes the issue.

tsender, 2024-02-04:
I apologize for causing this error. When I cleaned up the converters a while back, I must have overlooked those fields.