code-iai/ROSIntegration #208: Failed to build ROSIntegration on UE5 5.3.2 linux

ysdemiranda, 2024-03-24:
Hello,

I successfully built the ROSintegration plugin on Linux + Unreal Engine 5 by making a minor change to one of the source files.

For reference, this is one line of compilation output before making the change:

`In file included from /home/yasja/Documents/Code/unreal/ros_sophia/Plugins/ROSIntegration/Intermediate/Build/Linux/x64/UnrealEditor/Development/ROSIntegration/Module.ROSIntegration.cpp:122:
/home/yasja/Documents/Code/unreal/ros_sophia/Plugins/ROSIntegration/Source/ROSIntegration/Private/rosbridge2cpp/TCPConnection.cpp:28:48: error: overload resolution selected deleted operator '<<'
                std::cout << "Given IP address is invalid: " << *address << std::endl;`

Which blocked the plugin from compiling.

Changing the line mentioned from
`std::cout << "Given IP address is invalid: " << *address << std::endl;`
to
`std::cout << "Given IP address is invalid: " << TCHAR_TO_UTF8(*address) << std::endl;`

Fixed the issue.

Best Regards,

Yasja

aangert-anduril, 2024-05-31:
This fixed it for me too, it would be great to have a Pull Request for this feature!

aangert-anduril, 2024-06-05:
Merged to master here: https://github.com/code-iai/ROSIntegration/pull/210

Thanks for helping out @ysdemiranda

closed 2024-06-29 by ysdemiranda